prusa3d/PrusaSlicer #9254: Add shortcut button for default view

bretbailey1120, 2022-12-06:
When I bring a file into PrusaSlicer there is a default orientation that it originally shows. Then I can span and rotate around the object. I can also use shortcut number keys 1-6 to view the object at different angles. I'd like another shortcut key to get back to the original orientation. For instance Fusion360 has a "home" button to get back to their default orientation. Attached is a screenshot with the default orientation.

[image]

neophyl, 2022-12-06:
You might just want to try pressing the number zero.

bretbailey1120, 2022-12-06:
pressing zero works. Thanks!

closed 2022-12-06 by bretbailey1120