prusa3d/PrusaSlicer #9260: A pre-slice feature not showing up post-slice

jrgolz, 2022-12-07:
### Description of the bug

I've designed a utensil using Fusion 360. The last step of my design was to round off the edges at the back edge of the handle. When I bring this design to the platter, the design shows correctly. But after I slice, the design REVERTS to the design BEFORE I rounded off the two back corners. 

This is a very frustrating bug and I don't know how to resolve. 

### Project file & How to reproduce

[Utensil bug.zip](https://github.com/prusa3d/PrusaSlicer/files/10178883/Utensil.bug.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Win 11

### Printer model

Prusa mk3s

rtyr, 2022-12-07:
Please attach some screenshots what do you mean exactly.

jrgolz, 2022-12-07:
[image: image.png]
[image: image.png]
As you can see, the green platter image has rounded corners on the back of
the handle. In the orange post-sliced image, it ignores this edit. I know
it seems small, but the rounded corners are a very important feature of
this utensil. They are a must have for me. Do you know how to resolve this
issue?
Jason Golz
+1-415-279-4659

rtyr, 2022-12-07:
You have to attach the screenshots via web interface, it is not possible to use email attachements.

jrgolz, 2022-12-07:
[image]
[image]
 Here are the two images...

rtyr, 2022-12-08:
This is not suitable design for such big extrusion width / nozzle. Part of that sharp corner is not printable on such setup. Either modify the desin or use as small nozzle / extrusion width as possible.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]