prusa3d/PrusaSlicer #9261: "Delete All" should clear the saved project file name

jmsaltzman, 2022-12-08:
**Is your feature request related to a problem? Please describe.**

Previous project files can be accidentally overwritten after a scene is cleared. This can result in lost work.
After clearing the scene with "Edit > Delete All" (or Ctrl+Del, or the trash can button), the saved project file name is not cleared from the GUI memory. When the user loads a different piece of geometry and saves with "File > Save Project" (or Ctrl+S), the previous project file will be overwritten.  Similarly, when quitting the program, there is a "Closing PrusaSlicer. Current project is modified..." warning, and that should not come up when the scene has been cleared.

**Describe the solution you'd like**
Clear the current saved project file name variable in the GUI. It should be cleared in the same way the GUI doesn't know where to save a project file upon first launching the program with no geometry or project file.

**Describe how it would work**
The GUI's memory of the saved project file name should be cleared with "Delete All" just like the geometry.

**Describe alternatives you've considered**
I am in the habit of quitting and relaunching the app to avoid this issue.

**Additional context**
The GUI knows not to overwrite a project file with an empty scene: if a user tries to save a project file with no geometry loaded, the GUI will pop up a warning: "The plater is empty. Do you want to save the project?" Answering "Yes" would destroy work. The warning should not exist, instead a "Save as..." file browser should pop up.

Thank you!

foreachthing, 2022-12-10:
> "Delete All" should clear the saved project file name

No. "Delete all" is not the same as "new project".
I never use "delete all" but it's there for those who use it. So, just don't use it.

jmsaltzman, 2022-12-10:
The "Delete All" button is front and center and should be used. I use it. But it's an odd function: it clears the scene but not the project, which is a strange thing to do if a user is not intending to do something new.
It should be a shortcut to "New Project," really.

neophyl, 2022-12-11:
Its only odd to some people.  I happen to agree with foreachthing.  The functions are quite different.  As to what the shortcut should go to, again that is going to vary from person to person.  
Changing it would disrupt those who are used to the current workflow.

closed 2026-09-29 by prusaslicer-bot[bot]