prusa3d/PrusaSlicer #9263: First layer temperature set to other layer temperature at beginning of print.

Merlin246, 2022-12-09:
### Description of the bug

**Problem:** the "other layer" temperature is being used for the first layer instead of _first_layer_temperature._

**Cause:** extra gcode after custom start gcode causing the temperature to be set to "other layer" temperature instead of _first_layer_temperature_. See attached "GCODE.pdf", the highlighted line is causing the issue.

**Misc:** Running RepRap Firmware 3.4.3 on Duet 2 Wifi with Duex Expansion Board.

### Project file & How to reproduce

[Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/10190888/Issue.zip)
[GCODE.pdf](https://github.com/prusa3d/PrusaSlicer/files/10190916/GCODE.pdf)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11

### Printer model

E3D Toolchanger w Revo-Volcano-N/A-N/A

Merlin246, 2023-01-10:
> I see the issue. The problematic G10 g-code is not part of the Slicer custom g-codes, but rather emitted by the printer profile (toolchanger), and it seems to think it's not in the first layer anymore. It's difficult for me to reproduce without having an E3D Toolchanger though. I'll see if there's anything we can do.

Let me know if there is anything I can do to help (sharing the config, testing, etc). Would love to help get to the bottom of this!

Would it be easier to use an alternative to G10 in my custom startcode? Something like the [M568](https://docs.duet3d.com/en/User_manual/Reference/Gcodes#m104-set-extruder-temperature) gcode ([M104](https://docs.duet3d.com/User_manual/Reference/Gcodes#m109-set-extruder-temperature-and-wait) doesn't allow separate active & standby temperatures).

kubispe1, 2023-01-11:
Hi, in this case, PS Ooze prevention caused this issue. So the solution is to disable Ooze prevention. 
After that there is a G-Code preview bug, that was already fixed in 2.6.0. G-Code itself should be fine. 
So, I see this:
[image]

kubispe1, 2023-02-01:
I am closing this issue as solved with new alpha: 2.6.0-alpha2

closed 2023-02-01 by kubispe1

Holz167, 2024-07-18:
I see, thank you. Although, I'm not exactly seasoned user. What do I click where to get this fixed on my side? please