prusa3d/PrusaSlicer #9264: Error when multi-material printing with one extruder

PetrJan123, 2022-12-09:
### Description of the bug

Neumím anglicky tak používám automatický překladač, proto omluvte chyby v překladu, snad je u vás i česky mluvící vývojář.

V prusasliceru 2.5.0 je chyba při generování g-code multimateriálového tisku s jedním extruderem. Možná už tam je delší dobu, potřeboval jsem teď narychlo vytisknout na své Mk3s+ vícebarevné informační tabulky (dříve to šlo bez problémů) ale teď při prvních změnách barev proběhne výměna správně, ale v dalších vrstvách generuje příliš velké množství filamentu po změně barvy!
Vytvořil jsem si malinký model se třemi výměnami v jedné vrstvě abych mohl hledat v g-codu příčinu až jsem ji našel.Při výměnách v první vrstvě je hodnota extruze "F.8" v dalších vrstvách generuje slicer hodnotu "F4" a to je pětinásobek předešlé hodnoty, vytlačí tak na povrch bouli.
Chová se tak bez i s čistící věží.Přikládám obrázek s chybou.

Translate:
I don't speak English, so I use an automatic translator, so please excuse the mistakes in the translation, maybe you have a Czech-speaking developer.

There is a bug in prusaslicer 2.5.0 when generating g-code of multi-material printing with one extruder. Maybe it's been there for a long time, I just needed to quickly print multi-color information tables on my Mk3s+ (it used to work without problems) but now the first color changes will change correctly, but in the next layers it generates too much filament after the color change!
I created a tiny model with three exchanges in one layer so that I could look for the cause in g-code until I found it. For the exchanges in the first layer, the extrusion value is "F.8" in the other layers, the slicer generates the value "F4" and that is five times previous values, thus pushing the bulge to the surface.
It behaves like this without and with the cleaning tower. I am attaching a picture with the error.

### Project file & How to reproduce

[3barvy.zip](https://github.com/prusa3d/PrusaSlicer/files/10193536/3barvy.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0.+win64

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S

rtyr, 2022-12-09:
Na úvod jen zmíním, že volba "MultiMaterial tisk s jedním extruderem" není určena k účelu, k jakému ho používáte. Toto nastavení se používá pro MMU jednotku nebo podobný systém, kdy opravdu dochází z pohledu tiskárny/FW ke změně nástrojů. Toto s M600 je spíše takový hack, ač docela dost používaný.

Ty 4mm jsou proto, že tuto hodnotu máte nastavenou v nastavení extruderu při změně nástroje.  Jedná se tedy o očekávané chování.

[image]

PetrJan123, 2022-12-09:
Díky za informaci, jsem rád že je chyba na mé straně a ne přímo v PrusaSliceru.

V každém případě díky za práci co odvádíte pro podporu uživatelů 3D tisku.

closed 2022-12-09 by PetrJan123