prusa3d/PrusaSlicer #9266: pausing while printing and leaving zits

wbrokow1, 2022-12-09:
### Description of the bug

When I am printing I notice that the hot end pauses and leaves zits. 
I have tried slicing the same model with cura 14.07 and don't get the pausing at all.
also file and slicing, printing on bambu labs x1c with ams works fine.

delta kossel diy.
Smoothieboard 1.0a
smoothie firmware (tried 2 )

using prusa 2.5.0

### Project file & How to reproduce

[km5cyk2Part2.zip](https://github.com/prusa3d/PrusaSlicer/files/10197756/km5cyk2Part2.zip)
https://photos.google.com/photo/AF1QipM4-rn9b6hu80t8D9J-VANUvXYxWIeZ-_1Dp0Qu
[km5cyk2Part2 (2).zip](https://github.com/prusa3d/PrusaSlicer/files/10197763/km5cyk2Part2.2.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 11

### Printer model

delta kossel smoothieboard

murk-sy, 2022-12-12:
Can you post working Cura gcode? The provided gcode has 224 segments for each perimeter and your printer is likely having issues chewing through that gcode fast enough. Cura is either using G2/G3 commands (not sure if it has those?) or is simplifying the model when slicing.

wbrokow1, 2022-12-12:
[curacustom cylinder fancy bottom.zip](https://github.com/prusa3d/PrusaSlicer/files/10210471/curacustom.cylinder.fancy.bottom.zip)

rtyr, 2022-12-13:
You can try to change g-code resolution on advanced tab (higher value = lower g-code resolution). Overall it seems more like FW issue, since standard g-code works fine on Marlin/Klipper/RRF...

wbrokow1, 2022-12-13:
I changed the gcode res in prusa to .2 and it did not stop the pausing

wbrokow1, 2022-12-13:
ok changed slicing resolution to 0.01.
Still pausing

neophyl, 2022-12-13:
0.01 is a much lower number so more segments than 0.2  or was that a typo and you tried 0.1 ?

wbrokow1, 2022-12-13:
yes , slicing resolution to 0.01
gcode resolution to 0.2

mperino, 2022-12-14:
You may also want to try turning off "Power Loss Recovery" on the printer. There have been some timing issues where the "save" of the current state to the SD interferes with smooth layers because some printers/boards/firmware pause to write out the save state to the SD card.

wbrokow1, 2022-12-14:
There is no power recovery on smoothie.Thanks

wbrokow1, 2023-02-13:
Hi,
Sorry for the delay. Ok I tried with a coarse gcode resolution (0.1) and I got the same pausing. 
Please note I have tried 3 different smoothie FW.
I used CuRA 4.6.1 and the problem goes away.
Any other thoughts?

wbrokow1, 2023-05-03:
I figured this problem out.
In the firmware (smoothie) there is a setting called junction deviation.  This was set to a value which was extreme (0.5) 
proper setting for my printer was 0.1

wbrokow1, 2023-05-03:
solved. thanks for your input

closed 2023-05-03 by wbrokow1