prusa3d/PrusaSlicer #9267: Big shocks during the print of the low part of the support, loss of steps

walterbux, 2022-12-10:
### Description of the bug

Big shocks during the print of the low part of the support, loss of steps. If I use Curaengine to slice the same model with the same orientation and support no problem at all...

### Project file & How to reproduce

[mainsup.zip](https://github.com/prusa3d/PrusaSlicer/files/10199580/mainsup.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Ver. 2.5.0 + Win64

### Operating system

Windows 11

### Printer model

Prusa like custom printer

rtyr, 2022-12-13:
Big shocks of what? Loss steps of what motor? Extruder?

I am afraid we will be not able to troubleshoot issues with your printer here.

walterbux, 2022-12-13:
shocks on motion of X and or Y axes motors, loss of steps in X and Y
directions.
Thanks.

Walter Bux

closed 2026-09-29 by prusaslicer-bot[bot]