prusa3d/PrusaSlicer #9270: Changing not related part ignores extruder change

boristomas, 2022-12-10:
### Description of the bug

Changing first part ignores existence of the rest of parts

### Project file & How to reproduce

Normal behaviour: 
[image]

Faulty
[image]

First layer is the only relevant. Also notice that even normal behaviour is faulty: yellow is not shown (ignore material, I've set all to prusament petg and it is the sam)

Project file: [unutarnji107.zip](https://github.com/prusa3d/PrusaSlicer/files/10200013/unutarnji107.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.3.0+win64

### Operating system

Windows 10 x64

### Printer model

Prusa MK2 + MMU2

rtyr, 2022-12-13:
The model is not prepared correctly, the parts (letters) are just overlapping with the base. You have to either to create the model correctly (letters as independent parts substracted from the base) or change the type from "part" to "modifier" on all letter-parts.

closed 2022-12-13 by rtyr

boristomas, 2022-12-13:
@rtyr i disagree, i am using clipping feature
[image]

Nevertheless, your explanation doesn't explain faulty behaviour.

rtyr, 2022-12-13:
[image]

rtyr, 2022-12-13:
Just right-click on the gear icon and change the letters type to modifier, it should be enough.