prusa3d/PrusaSlicer #9271: Please add a function to convert from gcode back to 3mf [Feature Request]

laptop-laptop, 2022-12-11:
**Is your feature request related to a problem? Please describe.**

Accidently deleted the original 3mf file. Although I can open the corresponding gcode file from the gcode viewer in PrusaSlicer, there is no function to convert the gcode back to 3mf file.

**Describe the solution you'd like**
Somehow convert the gcode back to 3mf file.

cnlson, 2022-12-11:
yeah, that i am aware of there is no function to recreate a 3mf or even an stl from a gcode. If i was you I would try recovering the file. in the trash or even using file recovery software 
https://recoverit.wondershare.com/free-data-recovery/free-windows-10-data-recovery-software.html

foreachthing, 2022-12-11:
You can convert the GCode to .obj in prusa-gcodeviewer.exe.
[image]

But, reslice the sliced slices, doesn't slice properly. Sooooo.... I don't think you really can convert that in one easy step.
Spot the differences:
[image]

You could import the .obj to Blender and recreate the model to some degree.
Screenshot from Blender:
[image]

laptop-laptop, 2022-12-11:
Thanks for the interesting conversion procedure.

I have the original STL file. I just want to compare the settings I used for this project with those of another project.

neophyl, 2022-12-11:
That function to load in the settings used already exists.  File>Import Config and then select your gcode file.  This is possible because PS stores a copy of the profile settings at the end of the gcode file.  You can open up the gcode in any text editor and read them back manually if you like.

That will get you the settings in the print/filament/printer profiles.  It will NOT get back anything applied by modifiers or object specific settings.  Those are gone for good.  There isn't anything that will get those back.  Unless one of the normal file recovery methods will work as suggested by cnlson.    Like with ALL data for anything you should learn to do regular backups on anything that is important to you,

Your original feature request is markedly different to what you asked for in the last post.  Not the same thing at all.

rtyr, 2022-12-12:
I believe it is answered. Closing.

closed 2022-12-12 by rtyr