prusa3d/PrusaSlicer #9272: Lets eliminate hull line

sanczess, 2022-12-11:
So today the issue is over 100 years old and though all people know the source of it no one could reliably solve it yet.

Lets start a consideration.
A hull line is created when the printhead passes a sharp change of layertime. So you print print print and any time get a roof. 4 layers along you get a longer layertime than all the layers before. Not a big deal for the printer. But then the layertime bevomes suddenly much shorter and the pressure inside the bozzle is kind of unpredictible.
What happens now on my observation are 2 effects.
1. Perimeters shrink while printing the roof
2. After finishing the roof and start perimeters on new layer pressure in the nozzle is too high. 

So due to incorrect pressure inside the nozzle there appears a incorrect flow and molten plastic is pushed outside.

So if we know the source of the issue, why not solve it?
Ok I had one try in 2021 to solve it, but it was not really a solution but a reduction of the effect.

I modified the flat roof to a slightly slope surface. With that I eliminated a very sharp change of layertime. And yes, a hull line was no longer visible. 

But what if your model needs that flat roof with solid infill?
In my opinion now its time to modify the flow.

In order to create space where plastic is able to flow into due to the incorrect nozzle pressure I modified the gcode manually and reduced the flow on the 2 last outer perimeters. Of 4 perimeters in order from inner to outer the 3rd perimeter reduced to 50% and 4th to 90% flow.

Yes the result looks really way better but its not perfect.
Its a pain to modify the gcode manually and it needs to be done for 3 layers after the hull.

What we might need in first step is:
1. monitor layer time and maybe print time of one consistent contour if there are several in that layer.
2. Adjustment of flow for that case

Pictures following tomorrow

murk-sy, 2022-12-12:
The plastic is deposited (pretty much) exactly where it should be, but it doesn't stay there once cooled down - or the parts underneath don't. So no matter how you look at it, the issue is more or less related to plastic shrinkage (and probably internal stresses). Layer time is a consequence of the printing process, not so much the issue itself.

Shrinkage by itself is not *that* difficult to compensate for if you assume it's uniform, but it's not. Additionally, shrinkage of the currently printed printed layer will affect the layer directly below it, so you need to start compensating for full-layer shrinkage before you even print it.

Theoretically if plastic were cooled to ambient immediately after extruding, you would think that would solve the issue, but that's not the magic bullet either. To adhere the perimeter next to it or the layer above it, basically all already extruded plastic is heated again, which can release internal stresses and deform the shape.

However, the only reliable way to know what happens with the benchy hull line would be to run a print with a thermal camera and a high precision 3D scanner active at all times. Or, alternatively, do what high end SLS printer manufacturers do and run a thermal-physics simulation ahead of time and then pre-warp the printed object. However, then we're getting into the territory of controlled humidity and temperature of the filament and build chamber as well.

Even injection molding has stuff like sink marks, but there you can cheat in ways 3D printing cannot, for example by increasing injection pressure.

To be clear, I don't think it's an unsolvable issue - but I'm pretty sure no trick in current slicers can reasonably compensate for defects like it without dramatically affecting things like print speed or cooling.

bombela, 2022-12-14:
You could look at the pressure equalizer feature. This is a little bit related to the discussion. I use it to equalize the pressure from high flow to lower flow.

In effect it will slow down the last bit of infill (presumably printing at the highest flow) to equal the flowrate of the perimeter that comes next. I found this reduces some of the cosmetic uneven extrusion. This doesn't solve the varying layer time of course.

[image]

stepikovo, 2023-01-01:
@sanczess do you have the pictures promised in the first post? :)

stepikovo, 2023-01-01:
Linking #991  for future reference

MicroMan76, 2023-11-04:
(Also posted in a similar thread)

I've have this issue a lot, it's really causing me problems with my large parts. So I know the conditions under which it happens quite well.

I'm convinced it's got nothing to do with print settings and is either software(slicer) or a firmware issue because I'm able to produce it without the issue if I just adjust the size of the inside cut feature slightly. There seems to be a relationship between the distance of the flat face inside and the outer perimeter.

It also happens on my i3 and XL 5 head (any head).

This would then perhaps suggest that the tool path isn't accounting totally for the change in feature causing a tiny offset. I'm not saying that is the issue, but it could be a potential.

Here is a small and simple part I've created which produces the issue.

https://molyneuxdevelopments.com/benchy_hull_line_issue/

I'm really keen to resolve this problem as a lot of the parts I'm printing have this type of geometry.

20231104_120049

michaelhq, 2026-09-17:
I think this issue is still relevant and should remain open.

Shrink / hull lines are not simply a cosmetic artefact that can always be solved by tuning a single flow or pressure-advance value. They are often caused by abrupt changes in the thermal and mechanical conditions of a print, for example when sparse infill transitions to a large solid region or when a massive floor becomes connected to relatively thin outer walls.

There are also recent demonstrations showing that the effect can be reduced considerably by changing how these regions are connected or printed:

https://www.youtube.com/watch?v=d61u_lKFa-U  
https://www.youtube.com/watch?v=ITighzYPTTs

What makes this issue particularly interesting again is the new Lua plugin architecture introduced with PrusaSlicer 3.0.

The current Lua API is not yet sufficient for this, because plugins cannot inspect or modify the slicing pipeline. However, there is now a related discussion about exposing extension points for slicing decisions (#15746).

Shrink-line reduction would be a good real-world use case for such an API.

A plugin could, for example:

- detect abrupt transitions between sparse and solid regions across neighbouring layers,
- analyse changes in layer time and deposited material,
- detect strong coupling between large solid regions and external perimeters,
- adjust the ordering of perimeters and solid infill,
- locally modify infill/perimeter overlap or flow,
- or experimentally introduce more gradual transition strategies.

This would allow different compensation algorithms to be developed and tested without requiring every experimental strategy to become a built-in PrusaSlicer option.

For this to work properly, the plugin would need access before final G-code generation, while layer geometry and extrusion roles are still available. A pure G-code post-processing step would be too late for many of these approaches.

So I would suggest keeping #9272 open as the feature/problem issue, while #15746 can cover the more general API infrastructure that could eventually make experimental solutions possible.