prusa3d/PrusaSlicer #9274: sliced g-code name shows wrong setting for files sliced for MMU

Saschaka, 2022-12-12:
### Description of the bug

when you have different materials (e.g. PLA and PETG) in the 5 MMU slots, the generated g-Code follows the convention "project-name"_"layer height"_"material"_MK3SMMU2S_"time". the problem: if you slice something with PLA in slot 2-5 and slot 1 has PETG material configured in the slicer then the output file will be named "project-name"_"layer height"_"**PETG**"_MK3SMMU2S_"time". 

Is there a possibilty to have the name become only the material used and if you mix materials then it says remix or variable material or something?

### Project file & How to reproduce

[Transport-handle.zip](https://github.com/prusa3d/PrusaSlicer/files/10209509/Transport-handle.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 +win64

### Operating system

Windows 10

### Printer model

Prusa i3 Mk3s+ MMU2S

rtyr, 2022-12-13:
>  if you slice something with PLA in slot 2-5 and slot 1 has PETG material configured in the slicer then the output file will be named "project-name""layer height""PETG"MK3SMMU2S"time".

You are not using up-to-date print settings. We added new variable `initial_filament_type`, which will add the material, which is used at the start of the print, so PLA in this particular case.

Your output format should look like this:
`{input_filename_base}_{layer_height}mm_{initial_filament_type}_{printer_model}_{print_time}.gcode`

Saschaka, 2022-12-13:
Thanks! settings were indeed outdated

closed 2022-12-13 by rtyr