prusa3d/PrusaSlicer #9278: Multimaterial feature not slicing correctly

sango0711, 2022-12-13:
### Description of the bug

After coloring in a model with the multimaterial bucket feature, everything looks fine. But after slicing lines which should be i.e. white are partly brown. In red areas white spots apear.

### Project file & How to reproduce

Open reindeer2.3mf, slice, see preview. 

https://www.printables.com/de/model/339546-reindeer-ornament-mmu-single-extruder-version

raw stl included. 

[reindeer.zip](https://github.com/prusa3d/PrusaSlicer/files/10222275/reindeer.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Window 10

### Printer model

Prusa Mini+

sango0711, 2023-01-17:
In the preview it is fine. After slicing you see white spots in the gcode preview. In the zip reindeerbug2 shows the spots where it should not be white.

stepikovo, 2023-01-19:
Please send me a screenshot showing that issue

sango0711, 2023-01-19:
Thanks for helping out here. 
2 Things, i see you used the files from printables, which are the seperate files, which work fine. I used the single file, which can be colored in as well. Which is included in the zip file, not the 3mf from printables. Sorry for the confusion.

Then you get these previews. Coloring in the preview is good, but the slicing is going bad. I persume it is because the design goes through all layers, not just the top layer.
[image]

comm, 2023-11-13:
I am having a similar issue when using multimaterial painting:
[image]

cybergurke, 2023-11-21:
I have a strange error too.
[image]
[image]
[image]
[Home Pod TIE Fighter.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/13431465/Home.Pod.TIE.Fighter.3mf.zip)
Credits: https://www.printables.com/model/510395-star-wars-homepod-mini-stand-tie-fighter

Jan-Soustruznik, 2025-02-12:
Hello, closed as fixed as part of PrusaSlicer 2.9.0-alpha1 -  "The Improvements in painting tools and multimaterial segmentation."

For more details, please check the changelog: [PrusaSlicer 2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1).