prusa3d/PrusaSlicer #9281: Support expansion parameter

pavolNovak, 2022-12-14:
I print models from high temperature materials and need supports most of the times. I normaly use grid support since its going further than the part and the edge of the support interface is further away from the part. Downside is it uses more tima and more material. It would be great if I could adjust the support generation the same way how you can adjust the raft expansion

Snug and grid patter of support would have added parameter of expansion that would give you extra space for the material since the edge of the interface layer is most of the times thicker since the nozzle spends more time there , therefore I want to print on clean support interface rather than on the edges directly. 

added parameter of support expansion in mm

[image]
[image]

stepikovo, 2023-01-17:
Hi @pavolNovak, I'm not sure I understand your issue/request. If you choose the Grid style of supports, the support structure will expand beyond the edge of the model, isn't that what you want?
You can also "cheat" a little bit, by increasing the Pattern spacing to enlarge the expansion.

[image]

If I misunderstood, or your case is special and Grid doesn't work, please provide your printing project saved as 3MF.

pavolNovak, 2023-01-19:
hello @stepikovo not really a special case . Yes the grid you can "cheat" and make bigger pattern spacing. But if the pattern spacing is already dialed in and you cant change it, then its making things worse. In Snug the support is only directly under the part and would be good if the can expand the same way raft expansion. 
Not only its easier to peel supports off but also from the printing process they are more consistent ( changing direction of printing directly under the perimeters of the part.
[image]
you see on the picture those supports are already too close to the part. Without changinf the pattern spacing its not possible to expand it. But there on the bottom there is "First layer expansion" which is bigger. so the same principle would be great to apply here. I hope its more clear now. Thank you.

dp250f, 2023-02-12:
I would also like this feature as it would make Organic and Snug supports easier to remove when using a smaller z-gap. Gives you something to pry on/against or grab with pliers.

morphias2004, 2024-05-10:
This is a duplicate of the issue described in #1341 

Please upvote it there.

closed 2026-09-29 by prusaslicer-bot[bot]