prusa3d/PrusaSlicer #9283: [Feature request] Temporarily disable automatic slicing

cnlson, 2022-12-14:
**Is your feature request related to a problem? Please describe.**
when i am moving and modifying a lot it is a pain to have the slicer reslicing at each move. On the other hand when i am making few changes I appreciate it. 

**Describe the solution you'd like**
a check box in the object manipulation box to enable disable auto slicing. 

**Describe how it would work**
unchecking the box would temporarily disable auto slicing, rechecking would restart. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I have clicked stop repeatedly but i would rather be able to save the work. but i do appreciate auto slicing in 90% of the cases. 

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2022-12-14:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6893

cnlson, 2022-12-14:
Yes it does appear to be a duplicate of something that has not been assigned since 9-21. @bubnikv anything is acceptable to me, a keyboard shortcut would be awesome.

bombela, 2022-12-14:
You can use ctrl-r to reslice. This is what I do now. Since the "background" processing is fighting me constantly. Must be a hacked feature.

rtyr, 2022-12-15:
Closed as a duplicate.

closed 2022-12-15 by rtyr