prusa3d/PrusaSlicer #9284: [Feature Request] Show print time delta between slice setting changes.

jaddel, 2022-12-15:
When tweaking slicer settings, it is often important to know the resulting print time difference compared to the previous slice. To make this easier, I suggest adding a time delta view to the slicer interface, as shown in the screenshot below

[image]

It could also be very useful to display the time delta next to each slicer setting, along with a combined function to slice after each setting change.

0x6d64, 2023-01-03:
I highly suspect that displaying the time delta next to each setting will mean that the slicer has to run N times for N different checkboxes on a setting page, making this close to unusable if the model takes more than an eyeblink for each single slice.

But: I like to see the delta in the slice info, I do mental math for this constantly. What I would propose as an extension if this feature request:
* have a "set as baseline" button somewhere in the slicer (e.g. in the sliced info window). If the button gets pressed, all slicer settings get stored alongside the estimated time
* whenever a baseline is set, a difference _to the baseline_ set of settings gets displayed alongside the estimate (i.e. "estimated: 1h20m, -34m compared to baseline)
* the user can access a window where the differences in settings get displayed between current setting and baseline
* whenever the user wishes, they can set a new setting as "the baseline"

This would enable the user to understand how much time they saved with several small tweaks compared to their initial (default) settings.

jaddel, 2023-01-26:
The baseline, likea a timeline history when modeling in cad showing the changes that have been made sounds great.

Tupson444, 2026-09-12:
Interesting idea. Might as well display delta of grams used.