prusa3d/PrusaSlicer #9286: Prusa Profile Configuration 1.5

mizzouse, 2022-12-16:
### Description of the bug

Since configuration merge 1.5, my printer Mk3S+ with the latest firmware, drivers, and even configs/profiles has not been able to start a print without removing the start g-code "if filament_settings_id[initial_tool]=~/.*Prusament PA11.*/} section and resetting it back to 1.4 which is saved as a copy. From what I gather, filament_settings_id was only used on MK2.5 printers in firmware 2.2 versions. What happens to my printer is the z probe stops at the plate and never starts mesh bed leveling and then with octoprint, it disconnects.

### Project file & How to reproduce

No project file.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10 64bit

### Printer model

i3 MK3S+

rtyr, 2022-12-16:
> without removing the start g-code "if filament_settings_id[initial_tool]=~/.Prusament PA11./} section

I don't think it is related to your issue. It is just condition which changes intro line for one particular filament. The g-code output is basically the same as was before this change (unless Prusament PA is selected). It is also after MBL command, so it cannot affect MBL itself anyway. I guess it is some Octoprint related issue. You can try to print the file directly from SD card.

rtyr, 2023-02-01:
No response, not reported by anyone else. Closing.

closed 2023-02-01 by rtyr