prusa3d/PrusaSlicer #9287: Circular bridges should not follow the bridged gap

BenBE, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
When the slicer needs to bridge a circular gap (e.g. on top of a hollow torus) it often tries to follow the gap along that upper edge which causes unnecessarily long bridges, when zigzagging the bridge (with some adjustments) to cross perpendicular would be the much better path to achieve the bridge.

A simple test body to reproduce this situation is shown in the below OpenSCAD file:

```openscad
$fn = 360;

difference() {
    rotate_extrude(convexity = 10)
    translate([80, 0, 0])
    circle(r = 20);
    
    rotate_extrude(convexity = 10)
    translate([80, 0, 0])
    circle(r = 10);
    
    translate([-110, -110, -100]) {
        cube([220, 220, 100]);
    };
}
```

When trying to slice this, you will find locations where the slicer ends up going tangent on the torus, instead of radial, as demonstrated in this screenshot:
[image]

**Describe the solution you'd like**
**Describe how it would work**
When bridging a gap over an arc the slicer should try to build the bridge radial at each point, instead of picking just one alignment and sticking with it.

This could either be active for every bridge, or whenever the slicer detects that there exist two places on the bridged area where the minimal bridge distance spans an angle greater a given threshold (e.g. 10°).

Furthermore it might be an option to split such areas into smaller sections as long as the above condition is not met.

**Describe alternatives you've considered**
Both the "radial bridging" (always along the shortest path) and surface splitting should be viable options to tackle this issue. The bridge area subdivision approach might even be advantageous for more complex geometries where, when ordering bridges accordingly, even otherwise hard bridges may have a realistic chance of succeeding (e.g. where partial bridges can be used to suspend secondary bridges from).

sslupsky, 2024-05-23:
See also #7041

Jan-Soustruznik, 2024-05-28:
Issue closed as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7041