prusa3d/PrusaSlicer #9289: Support interface fan speed

jzagaja, 2022-12-17:
I'm unable to use high fan speed on supported layer. Auto fan cooling does not set it right and dissabling auto fan isn't a solution.

Super Slicer has it. 

[image]
[image]

xenon2008, 2023-05-03:
I'm switching from Superslicer back to PrusaSlicer because Superslicer has too many bugs for me ... :( and I don't feel like constantly troubleshooting anymore..
This function with the FAN in the support interface would be very helpful, precisely because it makes it much easier to remove the PETG supports, for example!
Please adopt this function in PrusaSlicer as well

gryphonB-2, 2023-05-10:
I'd like to support/confirm this issue, as per my duplicate issue: Cooling for support interface layers #10519
In it Tupson444 mentions that the g-code is commented with 'TYPE <layer type>' lines.
That at least allows writing a script to insert the proper g-code.

gryphonB-2, 2023-05-10:
A further consequence is that, where the part droops onto support at the support boundary, the part-support adhesion is maximal. This creates an unnecessary weak point in the my parts, as I favour slim, elegant parts. Removing the support risks breakage at exactly that edge.

yuranikspb, 2023-06-09:
Very cool function

habracoder, 2024-03-05:
After SuperSlicer, it really only one reson why I can't switch to the PrusaSlicer.

jzagaja, 2024-03-05:
Yes Prusa slicer sucks in cooling section.

Should be simple like:
overhangs angle this %
first layer above support this %
bridges this %
overall from % to %
auto within above on/off

Tupson444, 2026-09-12:
still relevant