prusa3d/PrusaSlicer #9291: Modifiers cut the perimeter

patrickleb, 2022-12-18:
### Description of the bug

Hello,
When using modifiers, if select 1 perimeter, the extenal shel ramain OK, but the inside of the part has no perimeter.
If using 2 perimeters, then the external shell is cutted and the whole modifier has 2 perimeters.
This happen when a modifier is in the part, those where are in the external of the part have not this problem.
there is a problem with the solid infill that is created instead the regular infill.

### Project file & How to reproduce

[prusaslicer-2.5-modifier_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/10253452/prusaslicer-2.5-modifier_issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64-202209060714

### Operating system

windows 10

### Printer model

Prusa Mk3s + bondtech

midnightsmith89, 2022-12-28:
Similar issue, the modifier cuts away the part no matter what you change 
[image]
[image]

[BrokenSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/10316160/BrokenSlicer.zip)

mikolaszuza, 2022-12-28:
Duplicate of [1143](https://github.com/prusa3d/PrusaSlicer/issues/1143). Closing.
"Currently neither PrusaSlicer nor the upstram Slic3r allow one to change parameters of just a piece of perimeter. The change is applied to a region, around which a perimeter is drawn. The only exception to this rule currently is a situation, where the override parameters modify the infill only. In that case PrusaSlicer is smart enough to generate single perimeter around the merged infill regions."

"We are aware of the not ideal current situation and we might rework it one day."

closed 2022-12-28 by mikolaszuza

patrickleb, 2022-12-29:
Ok, sorry, I had searched for a similar pb before posting, but couldn't find #1143.
Ok for the answer, thanks