prusa3d/PrusaSlicer #9292: Spiral Vase mode messes up rest of the print

hrvojebazina, 2022-12-18:
### Description of the bug

I have a model that's disc shaped with letters on it and then it has a tube on top of the disc. I want the tube to print in spiral vase mode, however when I turn on vase mode it messes up the disc part of the model. As you can see in the photos for whatever reason it doesn't print the whole layer under the letters, but instead it prints around the letters leaving a hole, and then it tries printing the letters on top of the hole. It happens regardless of the infill pattern.
[image]

[image]

However with vase mode off, it prints the layer under the letters normally and then the letters on top of it.
[image]

### Project file & How to reproduce

[Spiral Vase mode issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10254640/Spiral.Vase.mode.issue.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+MaxOS-arm64

### Operating system

macOS Ventura 13.1

### Printer model

Prusa Mini+

stepikovo, 2023-01-18:
Hello @hrvojebazina, I think this issue should be solved by #3204 therefore this one should be closed. What do you think?

hrvojebazina, 2023-01-19:
> Hello @hrvojebazina, I think this issue should be solved by #3204 therefore this one should be closed. What do you think?

That thread is 4 years old. Will the issue be addressed soon?

closed 2026-09-29 by prusaslicer-bot[bot]