prusa3d/PrusaSlicer #9294: prusaslicer-2.5.0 fails to build with boost-1.81

kobboi, 2022-12-18:
### Description of the bug

[329/364] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIM
E_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_L
INK -DBOOST_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBNEST2D_GEOMETR
IES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=9 -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_
TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0
/include -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0_build/src
/platform -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/hidapi/include -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/glib-2.0
 -I/usr/lib64/glib-2.0/include -I/usr/include/sysprof-4 -I/usr/lib64/libffi/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libmount -I/usr/include/blkid 
-I/usr/include/fribidi -I/usr/include/cairo -I/usr/include/lzo -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/gio-unix-2.0 -I/usr/in
clude/sysprof-3 -I/usr/include/atk-1.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0_buil
d/src/libslic3r -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/libnest2d/include -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlice
r-version_2.5.0/src/miniz -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/glu-libtess/include -isystem /usr/include/eigen3 -isystem /usr/lib64/w
x/include/gtk3-unicode-3.0-gtk3 -isystem /usr/include/wx-3.0-gtk3 -isystem /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/libigl -isystem /usr/in
clude/Imath  -DNDEBUG -O2 -march=znver3 -pipe -fomit-frame-pointer -g -ggdb3 -fext-numeric-literals -Wall -Wno-reorder -pthread -fPIC -fsigned-char -Werror=return-type -Wno-ignor
ed-attributes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHo
stDialogs.cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/slic3r/
GUI/PrintHostDialogs.cpp
FAILED: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -
DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOS
T_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBNEST2D_GEOMETRIES_libsli
c3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=9 -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXIN
TL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -
I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0_build/src/platform 
-I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/hidapi/include -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/glib-2.0 -I/usr/li
b64/glib-2.0/include -I/usr/include/sysprof-4 -I/usr/lib64/libffi/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libmount -I/usr/include/blkid -I/usr/inc
lude/fribidi -I/usr/include/cairo -I/usr/include/lzo -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/gio-unix-2.0 -I/usr/include/sysp
rof-3 -I/usr/include/atk-1.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0_build/src/libs
lic3r -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/libnest2d/include -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_
2.5.0/src/miniz -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/glu-libtess/include -isystem /usr/include/eigen3 -isystem /usr/lib64/wx/include/
gtk3-unicode-3.0-gtk3 -isystem /usr/include/wx-3.0-gtk3 -isystem /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/libigl -isystem /usr/include/Imat
h  -DNDEBUG -O2 -march=znver3 -pipe -fomit-frame-pointer -g -ggdb3 -fext-numeric-literals -Wall -Wno-reorder -pthread -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attribu
tes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.
cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/PrintHostDialogs.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/slic3r/GUI/PrintH
ostDialogs.cpp
In file included from /usr/include/boost/filesystem/path.hpp:21,
                 from /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/slic3r/GUI/PrintHostDialogs.hpp:6,
                 from /var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/slic3r/GUI/PrintHostDialogs.cpp:1:
/usr/include/boost/filesystem/detail/path_traits.hpp: In instantiation of ‘constexpr const bool boost::filesystem::detail::path_traits::is_convertible_to_path_source<const wxStri
ng&>::value’:
/usr/include/boost/type_traits/conjunction.hpp:31:27:   required from ‘struct boost::conjunction<boost::disjunction<boost::filesystem::detail::path_traits::is_path_source<const w
xString&>, boost::filesystem::detail::path_traits::is_convertible_to_path_source<const wxString&> >, boost::negation<boost::filesystem::detail::path_traits::is_native_path_source
<const wxString&> > >’
/usr/include/boost/filesystem/path.hpp:256:9:   required by substitution of ‘template<class Source, class> boost::filesystem::path::path(const Source&) [with Source = const wxStr
ing&; <template-parameter-1-2> = <missing>]’
/var/tmp/portage/media-gfx/prusaslicer-2.5.0/work/PrusaSlicer-version_2.5.0/src/slic3r/GUI/PrintHostDialogs.cpp:82:58:   required from here
/usr/include/boost/filesystem/detail/path_traits.hpp:495:85: error: call of overloaded ‘_check_convertible_to_path_source(const wxString&)’ is ambiguous
  495 |         sizeof(is_convertible_to_path_source< T >::_check_convertible_to_path_source(boost::declval< T const& >())) == sizeof(yes_type);
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/filesystem/detail/path_traits.hpp:480:21: note: candidate: ‘static boost::filesystem::detail::path_traits::yes_type boost::filesystem::detail::path_traits::is_convertible_to_path_source<T>::_check_convertible_to_path_source(const char*) [with T = const wxString&; boost::filesystem::detail::path_traits::yes_type = char]’
  480 |     static yes_type _check_convertible_to_path_source(const char*);
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/filesystem/detail/path_traits.hpp:481:21: note: candidate: ‘static boost::filesystem::detail::path_traits::yes_type boost::filesystem::detail::path_traits::is_convertible_to_path_source<T>::_check_convertible_to_path_source(const wchar_t*) [with T = const wxString&; boost::filesystem::detail::path_traits::yes_type = char]’
  481 |     static yes_type _check_convertible_to_path_source(const wchar_t*);
[... cut ...] ertible_to_path_source(...) [with T = const wxString&]’
  492 |     static no_type _check_convertible_to_path_source(...);
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

### Project file & How to reproduce

Trying to build on Gentoo (up-to-date ~amd64), after a boost upgrade from 1.80 to 1.81.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Gentoo Linux

### Printer model

( I own a Prusa i3 MK3)

kobboi, 2022-12-18:
Downstream bug report including full build log @ https://bugs.gentoo.org/887055

funkycode, 2022-12-23:
Can confirm that

bnavigator, 2023-01-31:
Downstream Gentoo's patch fixes it.

lukasmatena, 2024-02-19:
Closing, we are now on 1.83.

closed 2024-02-19 by lukasmatena