prusa3d/PrusaSlicer #9295: Presets and Build plate have been replaced/deleted

S1I7, 2022-12-19:
### Description of the bug

I reinstalled the program and it I am missing all of the material presets and the build plate looks different.

My slic3r currently looks like [this](https://imgur.com/a/TQpue7Z)

 

Before the build plate used to say Prusa Mk3s by josef prusa and there were presets for loads of materials, they are now gone. Ive tried uninstalling and reinstalling and nothing has fixed it.

 

Thanks for your help

### Project file & How to reproduce

[messedupmk3s.zip](https://github.com/prusa3d/PrusaSlicer/files/10256547/messedupmk3s.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.5.0+win64 Build: PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Operating System: Windows System Architecture: 64 bit Windows Version: Windows 10 (build 19042), 64-bit edition

### Printer model

Prusa Mk3s

neophyl, 2022-12-19:
Have you re-ran the wizard after reinstalling ?

S1I7, 2022-12-22:
It seems to work now thanks

Jan-Soustruznik, 2023-01-06:
Hello, @S1I7 , 

According to you last comment, problem has been already solved. 

May I close issue? 

Thank you, have a nice day

closed 2026-09-29 by prusaslicer-bot[bot]