prusa3d/PrusaSlicer #9296: After reinstalling Windows, all settings disappeared.

zmiter25, 2022-12-19:
### Description of the bug

How could the settings disappear from the portable version of the program? PrusaSlicer is located on the D:\Portable Soft\PrusaSlicer-2.5.0+win64-202209060714 drive should the settings be in this folder too? Or do you scatter them all over the PC? Then what's the point of a portable version?

### Project file & How to reproduce

Format system drive

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64-202209060714

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S

neophyl, 2022-12-19:
Didn’t think there was an official portable version?  
If you run the software normally the data is on your user account. You can always start the exe with a shortcut that has the —datadir location defined and then it uses the location as the app data store. That’s how I used to run it from a thumb drive.

lukasmatena, 2022-12-22:
@neophyl is right, there is no "portable version". PrusaSlicer configuration is not "scattered all over the PC", by default it is saved in ApplicationData folder, where most applications save it.

closed 2023-02-06 by lukasmatena