prusa3d/PrusaSlicer #9299: ChristmasOrnamentRandomFail

leroythedog, 2022-12-20:
### Description of the bug

I am trying to slice an stl of a christmas bell.  Slicer seems to randomly fill in the toop that is supposed to be open. 
I had thought I was using wrong settings, but I have discovered that Slicer just randomly slices it wrong.    I took screen shots of two different times.   I did not change any settings.  One time it was sliced ok, and next time it did not slice correctly.  

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0++MacOS-arm64

### Operating system

MacOS Monterey,  version12.6

### Printer model

Prusa 2.5s

stepikovo, 2023-01-18:
Hello @leroythedog, please provide the project file saved from PrusaSlicer as 3MF.

leroythedog, 2023-01-19:
Thanks, I will try the 3mf option.

closed 2026-09-29 by prusaslicer-bot[bot]