prusa3d/PrusaSlicer #9301: Fill bed with instances can cause toolpath to exceed print area when brim is enabled

davidhbrown, 2022-12-20:
### Description of the bug

If brim is enabled, the "Fill bed with instances" command will often cause the skirt to exceed the print area. It seems to create exactly the same number of objects in the same position whether brim is enabled or not, suggesting that it does not consider the additional space required for the brim.

I expect the "Fill bed with instances" to consider the width of the brim to ensure that the tool path remains in the print area. Possibly it already subtracts from the full bed dimensions to allow room for the skirt as I haven't seen this to be a problem except when brim is enabled. It should also consider the brim and perhaps rafts and supports would also cause this problem. 

[image]

### Project file & How to reproduce

[Shape-Cylinder.zip](https://github.com/prusa3d/PrusaSlicer/files/10270295/Shape-Cylinder.zip)

1. Start Prusa Slicer (in Platter tab/ 3D Editor view)
2. right click on print area; select Add Shape > Cylinder
3. enable Brim -- project file saved at this point
4. right-click cylinder; select Fill bed with instances
5. press tab or otherwise switch to Preview
6. observe reported ERROR: "A toolpath outside the print area was detected"

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 10 (build 19045), 64-bit edition

### Printer model

Original Prusa i3 MK3S+ (kit)

foreachthing, 2022-12-20:
If it would consider the brim , you'd be printing less instances.
So, disable brim (or adjust brim-distance), print and be happy 😉 
Imho: if I click "fill bed", it should disable brim to avoid this issue.

Also, my print bed is not rectanglar and PS has already enough issues with a custom bed-STL. Therefore I just disable brim and I'm ready to print.

davidhbrown, 2022-12-20:
@foreachthing, I appreciate that you're trying to be helpful and indeed, I will manually set a smaller number of instances and auto-arrange to work around this. But this is not the support forum. It's a place to report issues with the software.  

A brim has a purpose. The fill bed command most certainly should not disable it any more than it should enable it.

foreachthing, 2022-12-20:
I get your point. And I see how this is an issue to you. But, in my eyes, it really is not a big deal...
There are other contradicting settings in PS. And there are most likely more "buring" issues at hand. So, this issue could be a matter of support - please don't get this the wrong way! Sometimes a little change in a workflow will make life easier.

My ideal brim would not go around the parts, it would go along _one_ edge of my print bed.
My brim has only one purpose: Make sure there is filament in the nozzle ready to print. I am still tuning my start gcode so I won't need any brim anymore - ever.

davidhbrown, 2022-12-20:
Just the skirt can do it as well. Here, I changed that cylinder to 5mm before filling (which took a crazy long time). No brim, but just the standard 2mm skirt is out-of-area.
[image]

rkoptev, 2023-01-08:
I'm frequently experiencing the same issue. And i think it should be fixed in a way that takes into account skirt and brim since they are enabled by default. You don't even need to fill the whole print bed to cause this issue, take a look:
[image]
Default slicer settings:
[image]

lukasmatena, 2023-04-02:
Arrange function accounts for brim and skirt since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena

seti-ldn, 2023-04-07:
Wow @lukasmatena  I was so happy when I found this! I just started using 2.6.0 Alpha3 to slice a print job, and noticed that when I auto arranged my items, I ended up with a "toolpath outside of print area" error.  I thought "This is weird and annoying, I have to manually move my items"

I found this, thought "Amazing!" and downloaded Alpha6. Same issue. I have to manually move my items to make it print.  Attached screenshots of the "Auto arranged" layout that doesn't work, and the manually moved layout that does.  Skirt and brim settings are factory default for the 0.2mm QUALITY Mk3 printer profile.

[image]
[image]
[image]