prusa3d/PrusaSlicer #9309: big brims collides with waste tower on mmu prints by using auto arrange

Kyrunix, 2022-12-21:
### Description of the bug

Greetings everyone,
if you print a small multi color object with mmu2s, enable a big brim and use the auto arrange option, the brim can collide with waste tower.

Add multicolor object, in this case basic cube with a 15mm brim
[image]
Use auto arrange to arrange objects
[image]
Brim collides with waste tower basement
[image]
first layer view
[image]

This cube is just an example, because it happend by every object.

I think the "brim width"-parameter need to be added to auto arrange method. 

### Project file & How to reproduce

[brim-collision.zip](https://github.com/prusa3d/PrusaSlicer/files/10278360/brim-collision.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10 Professional

### Printer model

Mk3s+ with MMU2s

lukasmatena, 2023-04-02:
Arrange function accounts for brim and skirt since [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena

Snuff1eupagus, 2024-04-26:
It is already possible to control your brims, even though this individual knows how they have not shown this yet, however this may help.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s