prusa3d/PrusaSlicer #9310: how to dissable auto bed leveling

DiamondHead1022, 2022-12-21:
### Description of the bug

not 100% sure if this is a bug but every time i slice and print something for my creality cr 10 smart pro it automatiacly bed levels but that takes extra time which i dont want it to do 

### Project file & How to reproduce

slice the file then print it

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

win 10

### Printer model

creality cr-10 smart pro

neophyl, 2022-12-21:
That's controlled by whatever is configured in your start gcode.  There will be a line in there telling it to do the levelling.  If you don't want it, remove it.  
As you haven't attached the requested saved project file when opening an issue no one can check what its actually configured to do, only guess.

BigBen2003, 2022-12-23:
Auf dieser Webseite ist ein Artikel über die Druckbettnivilierung: https://the3dprinterbee.com/de/3d-drucker-druckbett-nivellierung/

Wenn ich ein Druck beginne,  werden liediglich 49 Punkte auf dem Druckbett angefahren und vermessen. Dieser Vorgang dauert nicht mal eine Minute.. Das ist nötig, da man nie die Druckbett-Platte extakt genau an die Stelle hin bekommt, wo es mal bei der letzten Kalibrierung war. Um ein gutes Druckergebnis zu bekommen,

closed 2026-09-29 by prusaslicer-bot[bot]