prusa3d/PrusaSlicer #9314: Retraction field in 2 places

zbuilder801, 2022-12-22:
### Description of the bug

Hello. My question is why are there 2 setting for Retraction with the same definition?
One setting is under Filament Settings/Retraction/Length, the other is under Printer Settings/Extruder 1/Length.
They can both be different values but what does each do and which takes precedence?
Thanks

[image]
[image]

### Project file & How to reproduce

[PulleyTest2.zip](https://github.com/prusa3d/PrusaSlicer/files/10289394/PulleyTest2.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.2+win64-202204251110

### Operating system

PrusaSlicer-2.4.2+win64-202204251110-64bit  Ram=18G

### Printer model

CR-10 Smart Pro

fontenele, 2022-12-22:
Hi @zbuilder801, in the option at filament panel you can override the retraction settings because this can be different for differents materials.

zbuilder801, 2022-12-22:
Are you saying  the Filament Settings tab/Retraction Length overrides the Printer settings?

rtyr, 2022-12-22:
> Are you saying the Filament Settings tab/Retraction Length overrides the Printer Settings tab?

Yes. Answered. Closing.

closed 2022-12-22 by rtyr