prusa3d/PrusaSlicer #9318: Exits sometimes while looking to import 3mf file

DiscusZ, 2022-12-22:
### Description of the bug

V 2.5.0 for win64  seems to just crash out and exit ( no error messages)  while scrolling to look for file to import.

File - > Import  -> Imprt STL/3mmf/step/obj/amf

file window pops open, while navigating in this window the slicer will completely exit no error message what so ever

### Project file & How to reproduce

No project file  involved  crashes while navigating file browser window

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0.win64

### Operating system

Windows 10

### Printer model

Eryone thinker S

DiscusZ, 2023-01-05:
No project file needs to be loaded  I am in the file window trying to load a progec file and scrolling through the files and then slicer just exits  no error nothing.  Also a screen shot wont help  nothing to screen shot

DiscusZ, 2023-01-05:
I think I know what it could be.   I turned off preview windows when scrolling stl files in a directory and slicer no longer exits with no warning

stepikovo, 2023-01-08:
Can it be just one corrupted STL file? Can you test that, please? Just put a batch of STL files in a separate folder and go through it via Slicer, then the next batch etc... And if you discover the problematic file, share it with us, please.

closed 2026-09-29 by prusaslicer-bot[bot]