prusa3d/PrusaSlicer #9322: Thumbnail preview with custom colors

pl4nkt0n, 2022-12-23:
**Is your feature request related to a problem? Please describe.**
Actual thumbnail color is always the same orange. With browser previews like in mainsail or octoprint you cant differ visually between the desired filaments, which the model was sliced for.

**Describe the solution you'd like**
I want be able to choose custom colors (for example a check-box with "reset to filament color" like the option with the extruder color).

**Describe how it would work**
i cant describe it on programmer level, but it should be similiar like the "choose extruder color" function.

**Describe alternatives you've considered**
As far i tried, there is no possibility to change thumbnail color within prusaslicer.

neophyl, 2022-12-23:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/4577

kubispe1, 2023-02-26:
Closing as a duplicate #4577

closed 2023-02-26 by kubispe1