prusa3d/PrusaSlicer #9323: Firmware setting - Autoprint the first/newest/only file on the SD card after inserting.

Valthol, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
I do believe that in 99% cases if we take out the SD card and insert it we ALWAYS want to print the first/newest file/only file on top of the list on the SD CARD.

**Describe the solution you'd like**
When you insert SD card it automatically goes to the browse menu, it would be awesome if a user could turn on an option in the settings that when SD card is inserted it automatically starts the print with newest file on it.

**Describe how it would work**
For one printer it is not an issue, for 10 printers and 3 times a day loading SD card with always a new file - it is pretty time saving!

**Describe alternatives you've considered**
Could be also that it will automatically start the print only if there is ONLY one file in the SD card.

neophyl, 2022-12-24:
Isn’t this an issue of the printer firmware and not of the slicer ?  If you have a Prusa printer then there are separate GitHub repositories for those printer firmwares.

Valthol, 2023-01-01:
Thank you, you are right! Closing the issue.

closed 2023-01-01 by Valthol