prusa3d/PrusaSlicer #9324: Output multiple gcode files after auto arrange sets more than one full bed of objects

Valthol, 2022-12-24:
**Is your feature request related to a problem? Please describe.**
When you auto arrange it creates few beds to the right if you have more object than can fit on one bed. For me it is always making like 5-8 beds, and then I just am slicing each of them, exporting, moving the whole new bed from the right to the original bed, slice, export, and repeat.

**Describe the solution you'd like**
Auto arrange ensures that the objects are placed inside strict boundries and should fit given original bed. If auto arrange prepared for me more beds than one it would be a nice feature to slice them and export all at once and have multiple final gcodes for each bed.

pavolNovak, 2023-01-02:
It would be a nice feature, Or at least to mke it work via Command line to make it faster

kubispe1, 2023-12-22:
I am closing this issue as duplicate. I know your FR is older but I would like to let open #11475 And I would like to ask you eventually discuss other features of Multiple Build plates in this thread #10704

closed 2023-12-22 by kubispe1