prusa3d/PrusaSlicer #9326: Feature Request: Two very simple settings that support multiple extruder systems to eliminate priming.

TobyCWood, 2022-12-24:
Prime Towers are **NOT** a solution. I can print without **ANY** priming at all with my Toolchanger and my IDEX Mendel TLD3. But a couple of simple things added to PS would be very helpful. The way to avoid priming is to simply retract the filament all the way out of the melt zone when the tool is parked. A couple of relatively simple settings added to PS would make this approach way easier!:

1. **Add Retraction speed to each extruder setting; "Retraction on Tool disabled".** For it to work correctly ALL the material has to be drawn up and when the retraction speed is too fast the small opening of the nozzle does not allow enough air to be drawn in causing a small amount of the molten material to break off in the nozzle. This drools out of the nozzle and can make for a small gap or blob at the start of the tool path. Retracting slowly allows the suction to be relieved via the opening of the nozzle rather than migrate to the other end at the entrance to the thermal barrier. This is what we see happening if drawn up too quickly. Adding this also allows one to retain the quicker, shorter retraction for within the tool path.

2. Allow each extruder to draw a settable skirt. Right now only one extruder can make a skirt. The skirt acts as an initial priming line which  brings the material to the tip of the nozzle. After it draws the skirt it then does the Retraction out of the melt zone the correct amount per the above settings. This is best done within the print gcode and not within the start gcode so that it conforms to the shape and area needs of the object being printed as well as allow the correct length during detraction.

TobyCWood, 2022-12-24:
Whoops! One more thing... Multiple extruder systems need to Z hop away from the print on a tool change. AKA lift on retraction. RRF does it on it's own via a tool change external script... BUT all other machines and methods need it in the slicer. S3D v5 got around that by making a Pre Toolchange and a Post toolchange gcode script. PS doesn't do it at all.

cbaugher, 2023-02-14:
Seconded.  IDEX printers are becoming more common and need specialized options.

awabom, 2023-07-06:
Thanks @TobyCWood for mentioning slow retraction on tool change - it seems to work great to help prevent oozing!

My current workaround until better IDEX-support is a G-code substitution that looks for e.g. G1 E-5 F2400 and replaces it with G1 E-5 F1200. I just hope no other retractions use that exact length...

Note also that the activation of the first tool after the 'start g-code' does not count as a tool change even if it is - so no tool-change retraction is done for that first switch (and therefore no unretraction later). My only solution for this was to write a python script combined with special logic in the start g-code, sadly :(

closed 2026-09-29 by prusaslicer-bot[bot]