prusa3d/PrusaSlicer #9327: Infill layer idea

DeForestDiamond, 2022-12-24:
Hey, 

I have an idea for you for prusa slicer, not sure if its a good one or not. say you had a 0.6mm nozzle, if you where doing 0.2mm layers would it be possible to do two perimeter layers and one infill at 0.4mm? The nozzel would still be level with the top perimeter layer so it wouldnt collide. That way you'd cut your infill time in half.

You all rock and thank you so much for the amazing slicer!

Thanks you again!!!!
Mike

DeForestDiamond, 2022-12-24:
Maybe a second idea along the same lines. 0.6mm nozzel. Do the exterior perimeters at 0.6 and over extruder on the interior ones to increase thickness and reduce layers to decrease time.

DeForestDiamond, 2022-12-24:
Oh my! Im on a roll here. Solid top and bottom infill could have the exposed surfaces at a lower layer height and extrusion width than the unexposed ones!

DeForestDiamond, 2022-12-24:
Interior and exterior perimeters could be different layer hights as well. That would should also increase strength to some degree by connecting the layers. Or, stagger perimeter layers by doing the first outside perimeter at 0.2 and the interior one at 0.1, then continue with all 0.2 so the layers are meshed together.

neophyl, 2022-12-24:
Your first suggestion the software already does. It’s one of the combining infill options.  I’m not at my pc otherwise I’d find the correct setting for you but it’s in there. 

If you want layers to print at different layer heights then use either a height range modifier or the variable layer height tool in the slicer. 

The other suggestions have already been made several times. Could you please do a search of both open and closed issues with as many relevant keywords please. Given that there are over 2400 open issues currently duplicates are not useful.

rtyr, 2022-12-24:
As neophyl mentioned, the first idea is already implemented for very long time (combine infill every x layer). Other mentioned ideas are indeed duplicates of existing issues. I am closing this.

closed 2022-12-24 by rtyr

DeForestDiamond, 2022-12-24:
Sorry, I tried to help.....