prusa3d/PrusaSlicer #9328: Slice-view widgets disappear from bottom of window

davidmcnabnz, 2022-12-24:
### Description of the bug

After slicing, a row of widgets appears, allowing choice of viewing options, including 'View' and 'Show' combo boxes.

But when moving/zooming/panning the sliced object canvas, this widgets row often just suddenly disappears, and I can't see any way to get it back. When it disappears, the 'View' combo box reverts to 'color' mode, and overrides any choices the user has made such as 'Feature Type'.

Any workaround or upcoming fix for this?

### Project file & How to reproduce

Slice a model. Notice a row of widgets appearing in the sliced view. Move and zoom and pan the sliced view around. Make changes in the slice height slider. Observe the row suddenly and unpredictably disappear.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Ubuntu 22.04 LTS

### Printer model

Creality CR-10 Max

Jan-Soustruznik, 2023-01-06:
Hello, @davidmcnabnz , 

Could you share a video of this issue. I cannot reproduce on my side.   

Thank you, have a nice day

davidmcnabnz, 2023-01-07:
Hi Jan-Soustruznik, attached is a screencast as requested. This demonstrates the post-slice gcode preview panel losing its setting widgets from the bottom of the screen, and changing the view mode from 'Feature Type' to 'Colour'.

https://user-images.githubusercontent.com/11789334/211169247-ecdaec07-933f-4ffb-bde7-b78811d3789c.mp4

Prusa 2.5.0 (AppImage), running on Ubuntu 22.04 LTS.

Cheers
David

davidmcnabnz, 2023-01-07:
Oops, for some reason Github thinks the prior video file is corrupted. Trying again with a version after reformatting with ffmpeg.

https://user-images.githubusercontent.com/11789334/211169367-34118fe2-50c7-47d7-96ae-8da9a47900d0.mp4

stepikovo, 2023-01-08:
@davidmcnabnz by dragging the vertical slider by the + sign and then releasing it (0:18 in the video) you just applied a colour change and therefore you have to reslice the model. You can see how _Export G-code_ changes to _Slice now_. To avoid that in the future, drag the slider by the "up arrow", not the + sign which has a function :)

closed 2023-01-08 by stepikovo

davidmcnabnz, 2023-01-08:
Great, thanks. 

This changes this issue from a possible 'software bug', to merely a UI/UX misunderstanding, which could be sorted as simply as tool-tipping the draggable widgets.

Cheers
David

Jan-Soustruznik, 2023-01-09:
Hello. @davidmcnabnz, 

We will try it perform in next release. By fix of this issue:https://github.com/prusa3d/PrusaSlicer/issues/8710. Project will be correctly marked as unsaved.

Thank you, have a nice day