prusa3d/PrusaSlicer #9331: Fan speed not 100% when limited by min. layer time

MarkMan0, 2022-12-25:
### Description of the bug

The description says:

> If estimated layer time is below ~15s, fan will run at 100% and print speed will be reduced so that no less than 15s are spent on that layer

This would suggest, that the gcode for the fan would be `M106 S255`, but this is not the case - the fan usually runs at 99% speed on the LCD. In the attached project, the gcode is `M106 S252.45` - exactly 99%. I noticed this, because my fan runs audibly faster at 100%, than it does at 99.

In the project, I set all speeds to 80mm/s, and disabled volumetric speed limits. The final speed is then limited by `slowdown_below_layer_time`, as expected, yet the fan is only set to 99% - it should be 100%. 

The speeds are set correctly, because if I scale the model to 200%, and set the infill to 90% the minimum layer time will be greater than 15s, so it will print at 80mm/s - verified on "Speed" screen. Fan in this case is running at 50%

### Project file & How to reproduce

[demo.zip](https://github.com/prusa3d/PrusaSlicer/files/10300571/demo.zip)

1. Try as is - fan speed is set to 99% on line 820, and remains till 112454
2. Scale object to 200%, set infill to 90% - fan is set to run at 50% until the end of the print

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 + win64

### Operating system

Windows 10

### Printer model

Ender 3 - customized

MarkMan0, 2023-01-19:
Also related to this issue is that printing speeds are inconsistent, when every layer is the same shape - consider a box. Again, the speed is limited by the layer time from cooling. The STL was exported from F360 on high resolution, and the walls are simply extruded - basic shapes.
[image]
[Box.zip](https://github.com/prusa3d/PrusaSlicer/files/10455339/Box.zip)

bubnikv, 2023-06-13:
What's your operating system? Is it macOS Ventura?

closed 2026-09-29 by prusaslicer-bot[bot]