prusa3d/PrusaSlicer #9333: Jumps between lines without anchors

zero5, 2022-12-27:
### Description of the bug

hi!
I'm facing a problem with rectilinear infill when printing without anchors. In this mode it jumps between lines while i was expecting to see sequential travel of the extruder. It seems like unrational move but probably i'm not getting the reason. I'm new to the slicer so might not getting it right. Thanks in advance!
[image]

### Project file & How to reproduce

[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/10306662/Shape-Box.zip)
Just slice it and move line by line 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.5.0+win64-202209060714

### Operating system

Windows 11

### Printer model

no printer yet

zero5, 2023-01-10:
I've found out that making path chaining for polylines and excluding extrusion ordering and chaining made things behave like i expected. Not sure if it's a good way to do it but worked for my task.

bubnikv, 2023-01-10:
>  I've found out that making path chaining for polylines and excluding
extrusion ordering and chaining made things behave like i expected. Not
sure if it's a good way to do it but worked for my task.

I don't understand your suggestion. Do you have a pull  request? What your
task are you talking about?

zero5, 2023-01-10:
It was just my guess but not sure if it's needed. Probably would try to make pr later. And my task was to make extruder move sequentially between closest lines but not jumping around whole model. I suppose method for extrusion chaining and ordering works not properly but i'm not sure yet how it sould be.

closed 2026-09-29 by prusaslicer-bot[bot]