prusa3d/PrusaSlicer #9334: Crash with "Complete individual objects"

xarbit, 2022-12-27:
### Description of the bug

Please see the main Issue on the FlatHub project:
https://github.com/flathub/com.prusa3d.PrusaSlicer/issues/55#issuecomment-1365623343

Here is a patch from @eliadevito, it might be of interest for Prusa3D too: 
https://github.com/flathub/com.prusa3d.PrusaSlicer/blob/deaa297c37975a3aef327a0c539e5928aaaf8f33/patches/fix_55.patch

If so, please let us know so we can decide what to to with it.

### Project file & How to reproduce

Attached 3mf to reproduce.
https://github.com/flathub/com.prusa3d.PrusaSlicer/files/10078622/onder.zip

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 from flathub

### Operating system

Linux

### Printer model

MK3S+/Mini

Jan-Soustruznik, 2023-01-09:
Hello, @xarbit 

Thank you for report, but I cannot reproduce mentioned crash on PrusaSlicer. Could you provide more detail, please? -  Steps to reproduce.

Thank you, have a nice day

eliadevito, 2023-01-09:
Hi to all,
I found that the change applied by my patch is already present on the master branch, it is part of this commit https://github.com/prusa3d/PrusaSlicer/pull/7854/commits/90976c83d1c18dee141b05186e4821359b91c5af that  fix another issue.

At this point I think this can be closed and that my patch can be merged on flatpak package until the release of the next version of PrusaSlicer

closed 2023-01-09 by xarbit