prusa3d/PrusaSlicer #9337: Make "Max volumetric speed" a setting in Printer Settings > Machine limits

lukasradek, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
My machine is limited by a weak hotend so I was looking for some option to limit the printspeed by extruder feedrate.
At first I was looking for something like "Max hotend feedrate", which would be the same as "Max extruder feedrate" but taking hotend capabilities into account and limiting print speed by that.
Then I found "Max volumentric speed" inside Print settings, which is exactly that (except in mm3/s).

**Describe the solution you'd like**
I think it would be more logical to place "Max volumetric speed" in Printer settings since it is basically a machine limit.

**Describe alternatives you've considered**
I would consider either putting it under the "Machine limits" tab or possible inside individual Extruder settings, since it can be tool dependent in toolchanger machines.

neophyl, 2022-12-29:
It’s also effected by the filament which is why there’s one in the filament too, the lowest value configured is what is used I think.

lukasradek, 2022-12-29:
True... great adept for a machine limit with filament override.

Mjankor, 2023-01-01:
Yes please. It would make management of multiple machines, especially for those with a mix of high flow and standard hotends, much easier.

Ideally, max_volumetric_speed would be under the extruder portion of the Printer Settings.
And, filament_max_volumetric speed would accept percentages of the max_volumetric_speed (e.g., 50% for PETG)

Then each machine could have a max volumetric speed derived from their hotend capability, with a second, percentage based, limiter to account for filament types.

Mjankor, 2023-01-01:
Note, very similar to #7604

closed 2026-09-29 by prusaslicer-bot[bot]