prusa3d/PrusaSlicer #9338: Way to turn off automatic slicing

49erNutz, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
Whenever I try to add, move, duplicate, or organize, I have to wait until it finishes slicing. This is very annoying and time-consuming.

**Describe the solution you'd like**
Please provide a choice in the Preferences of auto slicing being turned on with it off by default.
Please place a start button within the slicer bar at the bottom right.

**Describe how it would work**
This would allow the user to work uninterrupted while preparing the burn plate. 

**Describe alternatives you've considered**
Turning on auto slicing after the initial slice, then turning it off after the slice has been exported.

**Additional context**
Thank You

DAflamingFOX, 2022-12-29:
This is already possible. Configuration -> Preferences -> General -> Background processing

stepikovo, 2023-01-03:
Exactly as @DAflamingFOX said, this already exists. Closing.

closed 2023-01-03 by stepikovo

hovissimo, 2024-09-06:
May I suggest a change to the name of the option in the settings?  I found that setting, but assumed it meant slicing in a different thread or process (and that it was broken, because the UI is clearly blocked during slicing). I had to search the web to find this issue to understand I had already found and dismissed the setting I wanted.