prusa3d/PrusaSlicer #9340: Enable height range modifiers in vase mode.

yellowcub, 2022-12-30:
**Is your feature request related to a problem? Please describe.**
The use of height range modifiers (of any type) in vase mode is disabled since PrusaSlicer equates it to a filament change process.  Encountered problem while attempting to make vase mode tests for speed and temperature.

**Describe the solution you'd like**
When using range height modifiers in vase mode, check if modifier is truly incompatible with vase mode.

**Describe how it would work**
Algorithm would check beyond height range modifier to see if implementation is incompatible with print mode.

**Describe alternatives you've considered**
None

**Additional context**
None

kubispe1, 2023-01-18:
Hi, maybe I have missed the point. But the Height range modifiers are available in spiral mode... 
See attached 3mf.
[Shape-Cylinder_9340.zip](https://github.com/prusa3d/PrusaSlicer/files/10440615/Shape-Cylinder_9340.zip)
[image]

yellowcub, 2023-01-22:
The height range modifiers work by themselves, but if you add a modifier to them - such as speed - PrisaSlicer gives the error "The Spiral Vase option can only be used when printing single material objects."
[image]

blownupp, 2023-08-14:
Looks like this is still happening in 2.6.0

[image]

memyselfandm, 2024-02-05:
Still an issue in 2.7.1. I'm trying to apply speed changes to the bottom layers (to slow down and ensure good adhesion and base) while the rest of the build is at much higher speed.

scottyallen, 2024-04-28:
Still an issue in 2.7.4.  At the very least, the error message should be more accurate.  I tried changing speed as well (which, afaict, shouldn't be an issue in vase mode), and the error "The Spiral Vase option can only be used when printing single material objects" is pretty confusing.

unixunion, 2024-07-07:
This is still an issue with 2.8.0. Please fix this as taller models in vase mode should allow me to adjust speed in the slicer so i dont have to manually tweak it for specific geometry areas.

closed 2026-09-29 by prusaslicer-bot[bot]