prusa3d/PrusaSlicer #9342: Layer by layer gcode view

Wikunia, 2022-12-30:
**Is your feature request related to a problem? Please describe.**
It would be great to have a layer by layer view option in the gcode viewer. Sometimes it makes sense to see a single layer where layer is defined by a number and not the printing z height. For example some parts might be printed a bit higher or lower in each layer but are conceptually still the same layer. 

**Describe the solution you'd like**
Would be awesome to have a switch between the current z view and a layer view. In the layer view it should still be a 3d view such that the lines are at the actual z position though just moving through the layers would be way easier.

**Describe how it would work**
Take the `;LAYER_CHANGE` or `;LAYER` comment into account for the ruler and have switch in the bottom right or inside the setting menu in the bottom right.

gordo3di, 2022-12-30:
Unless I'm confused, under preferences you can check box (in the image below) then you can scrub through just the current top layer instead of the whole gcode file. Also under the layer preview slider, you can pull up the bottom orange arrow so it will show just a range of layers, or a single layer, instead of the whole gcode file. The number in parentheses is the layer number.

[image]

[image]

Wikunia, 2022-12-30:
The problem happens if you have different Z values in one layer. For example print one part of the layer at a slightly different height. Then currently the slider has two steps for a single layer.

gordo3di, 2022-12-30:
Gotcha. I noticed that with the dense support layer.

closed 2026-09-29 by prusaslicer-bot[bot]