prusa3d/PrusaSlicer #9345: cmd+numbers shortcuts not working in MacOS, Czech keyboard layout

Stooovie, 2022-12-31:
### Description of the bug

cmd+1,2,3... shortcuts do not work in MacOS with Czech keyboard layout (nothing happens). They do work when I switch to US layout. 

### Project file & How to reproduce

Switch keyboard layout to Czech and press any cmd+alphanumeric number shortcut.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-x64

### Operating system

MacOS 12.6.2

### Printer model

Ender 3

lukasmatena, 2023-02-01:
Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

Stooovie, 2023-02-01:
Still not working in alpha 2... Monterey 12.6.3.

Stooovie, 2023-02-04:
Unfixed as of alpha3

Stooovie, 2023-09-04:
Unfixed as of 2.6.1  rc2

Stooovie, 2023-09-04:
> Fixed in [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2). Closing.

no, not fixed as of 2.6.1 rc2, never fixed in any release in between

Stooovie, 2023-10-22:
Still unfixed as of 2.7.0 alpha

lukasmatena, 2023-10-23:
We will investigate the issue again.

Stooovie, 2023-11-24:
Still unfixed as of 2.7.0.

Stooovie, 2023-12-17:
Still unfixed as of 2.7.1.

Stooovie, 2024-07-18:
Still unfixed as of 2.8.

Stooovie, 2024-07-18:
> We will investigate the issue again.

Well, did you?

closed 2026-09-29 by prusaslicer-bot[bot]