prusa3d/PrusaSlicer #9348: Support more dedicated Overhang perimeter settings

meltiseugen, 2023-01-01:
It would be nice to have some dedicated overhang perimeter settings like:

* Set an angle over which the perimeters are treated like overhangs (similar to Cura). Currently, only perimeters that are completely in the air are considered overhangs (bridge settings)
* Dedicated flow
* Dedicated speed

meltiseugen, 2023-01-09:
Anyone from the development team got an opinion on this? :D

kubispe1, 2023-01-09:
Hi, thanks for the request. We are working on some overhang enhancements for next release.

meltiseugen, 2023-01-09:
> Hi, thanks for the request. We are working on some overhang enhancements for next release.

Meaning 2.6? :O

lukasmatena, 2023-02-01:
Please, check out the new alpha: [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2) and let us know what you think about "Dynamic overhang speed". Thanks.

eugenmeltis3pg, 2023-02-03:
> Please, check out the new alpha: [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2) and let us know what you think about "Dynamic overhang speed". Thanks.

I will check today. Thank you for looking into this!

eugenmeltis3pg, 2023-02-03:
> Please, check out the new alpha: [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2) and let us know what you think about "Dynamic overhang speed". Thanks.

This is so much more than I could have hoped for. And It works with pressure equalizer as well!! 

Only think I noticed is that in the release notes the information is not complete
[image]

meltiseugen, 2023-02-03:
> Please, check out the new alpha: [2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2) and let us know what you think about "Dynamic overhang speed". Thanks.

This is so much more than I could have hoped for. And It works with pressure equalizer as well!!

Only think I noticed is that in the release notes, the information is not complete.
[image]

lukasmatena, 2023-02-03:
> Only think I noticed is that in the release notes, the information is not complete.

It is complete. That is one of recent github "improvements" - they are shortening the release logs in the preview. You have to scroll to the top of the release log and click the title to open just the release log of 2.6.0-alpha2. If you asked me, it is extremely stupid idea to only show "..." instead of writing "the preview is shortened, click here to open in full", but all I can do is complain. I will, in fact.

meltiseugen, 2023-02-03:
Loving the answer. Yes, complain :))

closed 2023-02-03 by meltiseugen