prusa3d/PrusaSlicer #9349: nefunguje rozdělení na objekty

Drobek-Petr, 2023-01-01:
### Description of the bug

Chtěl jsem stl rozdělit na objekty. Po několika hodinové práci mého pc se ukázala chyba o nedostatku ram paměti a program se ukončí. K rozděleným objektům se tak není možné dostat.
[image]
[Hrdlo 1_0.3mm_PETG_MK3S_15h20m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/10329306/Hrdlo.1_0.3mm_PETG_MK3S_15h20m.gcode.zip)

### Project file & How to reproduce

[Hrdlo 1_0.3mm_PETG_MK3S_15h20m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/10329307/Hrdlo.1_0.3mm_PETG_MK3S_15h20m.gcode.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Verze 2.5.0+win64

### Operating system

Windows 10 Enterprise Professiona

### Printer model

prusa i3 MKs3

neophyl, 2023-01-02:
Gcode is not a project file.  The request when opening an issue is for a saved Prusa Slicer project file, File>Save Project As.  That will save a .3mf file.  That along with a step by step guide to reproduce the issue from that saved project point is the best/only way that the devs have any chance of reproducing any problem.

Jan-Soustruznik, 2023-01-09:
Dobrý den, pane Kudera, 

.3mf soubor by měl být menší než samotné .stl, protože jsou data zabalena podobně jako .zip. Ohledně generování .stl z g-codu, nejsem si jistý, zda to slicer umí. G-cod soubor obsahuje instrukce pro tiskárnu - jedná se tak o finální produkt sliceru, který slicer nedokáže zpětně převést.

closed 2026-09-29 by prusaslicer-bot[bot]