prusa3d/PrusaSlicer #9350: Hitting Slice button closes application

BRoadifer, 2023-01-02:
### Description of the bug

This just started when I upgraded to 2.5.  I add a model, doesn't matter if its a simple or more complex model... As soon as I hit the slice button it crashes the slicer.  I also teste this with the Bambu Slicer based on Prusa Slice and it completely rebooted my machine when hitting the slice button.

Intel i7 12700k
GPU: Nvidia 2080ti

### Project file & How to reproduce

[tv-playfield_mount.zip](https://github.com/prusa3d/PrusaSlicer/files/10332524/tv-playfield_mount.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

Win 10

### Printer model

PrusaMK3S+ MMUv2

BigBrummbar, 2023-01-03:
I have this exact same option. A super basic model crashes my Prusalicer no matter what options I pick. 

AMD Ryzen 9 5900X
Asus RTX 3080TI

BRoadifer, 2023-01-04:
I have downgraded back to PrusaSlicer-2.4.2+win64-202204251110 and it does not crash when hitting slice button.  This seems to be caused by something introduced by v2.5.

BigBrummbar, 2023-01-04:
Understood. I'll find a download link for 2.4.2 then. Thanks for the follow up

BigBrummbar, 2023-01-05:
Yup. 2.4.2 didn't fix it at all. Perhaps I have to remove the program and all drivers for Prusa entirely to make it work? I did get an error when installing 2.4.2 that the current config version wouldn't work with 2.4.2 so maybe that's it? 

Regardless, Still can't slice models.

kubispe1, 2023-01-08:
Hi, guys sad to hear that the slicing caused you a crash. 
I cannot reproduce it. I sliced with succed:
2.5.0.
2.6.0.
even BambuStudio does not fail :D (import only geometry) 
Let's see if someone else could provide more details
 Windows 10

lukasmatena, 2024-02-19:
Closing, does not look like a problem in PrusaSlicer.

closed 2024-02-19 by lukasmatena