prusa3d/PrusaSlicer #9353: Paintable Fuzzy skin

blankcode, 2023-01-04:
I have a shower basket model I've modeled, and I'd like to designate a single face of the model to have fuzzy skin.

Like support enforcers, I'd like to be able to paint where I want Fuzzy skin. (Also, Ironing would be helpful in some cases.)

Select the tool and paint where you want fuzzy skin or paint to block fuzzy skin. (just like Support enforcers.)

I considered making a block the size of the space I want to be fuzzy, but it's going to alter the dimensions of the model slightly.
I considered making my model fuzzy, but FreeCAD is sluggish enough with models like this as it is.

Blue: Fuzzy Skin
Gray: Not Fuzzy
Red: MUST NOT be Fuzzy
[image]

neophyl, 2023-01-04:
Isnt this just a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6988 (and several others) ?

0x6d64, 2023-01-11:
I like the proposal, but just FYI: The desired result can already be achieved right now using a modifier body (both for fuzzy skin and ironing). Add a modifier box that intersects with the wall i the blue area (pay attention that it does not stick inside of the box), tell the modifier that it should enable fuzzy skin, done.

kubispe1, 2023-01-12:
Thanks, guys. Closing as a duplicate

closed 2023-01-12 by kubispe1

DVSVIDEO, 2023-02-22:
It's a very coveted request (from me too), so it might be worth addressing...
especially since the current solution is flawed
Thanks in advance.