prusa3d/PrusaSlicer #9354: MMU Printing isue- no_wipe_tower with single color change - don't load next filament correctly.

grabolukasz, 2023-01-04:
### Description of the bug

[image]
I have a problem during printing using multilateral. I don't need to use a wipe tower for this simple model. But when there is a problem when filament change is about to happen there the problem occurs. The printer loads the filament but just to the filament sensor position, not inserting it in the hotend till its end so that the filament cannot reach the nozzle through two layers.

And this is what is happening when printing, after three layers of filament change (start new color printing in air):
[image]

[cechownia_magnesik_v2_0.2mm_PLA_MK3SMMU2S_26m.zip](https://github.com/prusa3d/PrusaSlicer/files/10342703/cechownia_magnesik_v2_0.2mm_PLA_MK3SMMU2S_26m.zip)

### Project file & How to reproduce

Project file for printer:

[cechownia_magnesik_v2_0.2mm_PLA_MK3SMMU2S_26m.zip](https://github.com/prusa3d/PrusaSlicer/files/10342696/cechownia_magnesik_v2_0.2mm_PLA_MK3SMMU2S_26m.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10

### Printer model

Prusa I3 MK3S with MMU 2.0

Jan-Soustruznik, 2023-01-06:
Hello, @grabolukasz, 

for printing with MMU you have to always use the wipe tower. In this case you can use for material consumption reduction function - **No sparse layer(EXPERIMENTAL)** (_Print Settings -> Multiple Extruders -> No sparse layers(EXPERIMENTAL)_). If enabled, the wipe tower will not be printed on layers with no tool changes.

May I close the issue? 

Thank you, have a nice day

grabolukasz, 2023-01-13:
Hello,
As i mentioned, when i use option  *No sparse layer(EXPERIMENTAL), *this
not fill nozzle with filament, so it's causing a gap between layers,
because there is a gap in extruder between two filaments, previous and
loaded one.
So there is no solution for this.
This option is not working.
With greetins,
*Łukasz  *

Jan-Soustruznik, 2023-01-16:
Hello, @grabolukasz,

Could you share the .3mf file, please? 

Thank you, have a nice day

grabolukasz, 2023-01-17:
Hi,
I attach the 3mf File,
Have a nice day
Łukasz

petrubecheru, 2023-02-04:
for the gap between layers, see bug #9592

grabolukasz, 2023-02-05:
Hi,
Thank you, i know this issue, i foud solutions. don't use experimental
setting for wipe tower (no_sprase_layers), and you can print without lack
of layers, but you are using too much filament for wipe tower.
For 2.6 Alpha 2 there is the same problem.
Is there any solution for this issue?

With greetings,
Łukasz Grabowski

Jan-Soustruznik, 2023-02-16:
Hello, @grabolukasz, 

As I mentioned in my first comment ... without wipe tower isn't possible print. It is fact that in the nozzle is insufficient pressure, so that is your issue why is missing layer after color change. 

If you want reduce the material consumption use: **No sparse layer(EXPERIMENTAL)** (Print Settings -> Multiple Extruders -> No sparse layers(EXPERIMENTAL)). If enabled, the wipe tower will not be printed on layers with no tool changes.

I close issue as solved. 

Thank you, have a nice day

closed 2023-02-16 by Jan-Soustruznik