prusa3d/PrusaSlicer #9357: Outer perimeter leaves the object in different hights

ReinAir1511, 2023-01-04:
### Description of the bug

Outer perimeter leaves the object in different hights. Problem could be solved by rotating the object in Z.

[image]

[image]

[image]

### Project file & How to reproduce

That is the file including the project:

[FM_Montage.zip](https://github.com/prusa3d/PrusaSlicer/files/10347333/FM_Montage.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Windows 11

### Printer model

Prusa MK2S

Jan-Soustruznik, 2023-01-06:
Hello, @ReinAir1511, 

Thank you for report, but it seems that the issue is already reported -  https://github.com/prusa3d/PrusaSlicer/issues/8846. We will try to solve in next release. Closed as duplicate.

Thank you, have a nice day

closed 2023-01-06 by Jan-Soustruznik

kubispe1, 2023-01-08:
Thanks to both of you, looks like the Voronoi diagram generator issue.