prusa3d/PrusaSlicer #9360: CLI-processing throws "Objects could not fit on the bed" if object is far away

el-bart, 2023-01-05:
### Description of the bug

model, that's small, but positioned far away from center of coordinates, can be placed on a print plate in GUI. however when slicing from command line, program core-dumps with error that object does not fill the bed:

```
$ prusa-slicer \
  --loglevel 5 \
  --export-gcode \
  --output out.gcode \
  input.stl
[2023-01-05 21:27:24.276840] [0x00007f0399e8e900] [trace]   Initializing StaticPrintConfigs
[2023-01-05 21:27:24.292581] [0x00007f0399e8e900] [debug]   TriangleMesh::repair() started
[2023-01-05 21:27:24.293499] [0x00007f0399e8e900] [debug]   TriangleMesh::repair() finished
terminate called after throwing an instance of 'Slic3r::RuntimeError'
  what():  Objects could not fit on the bed
Aborted (core dumped)
```

[OpenSCAD](https://openscad.org) source for "faulty" object:
```
translate(10*1000*[1,1,1])
  cube(10*[1,1,1]);
```

the same object, but with smaller "translation" slices as expected, eg:
```
translate(10*[1,1,1])
  cube(10*[1,1,1]);
```
works fine.

### Project file & How to reproduce

source files: [problem.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/10355798/problem.tar.gz)
to reproduce unpack call `./run`.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

linux, debian 11

### Printer model

anycubic chiron

kubispe1, 2023-01-12:
I played with CLI (on WIN 10) and the behaviour is quite strange in my eyes. 
I reach the most versatile and flexible result like this:
`prusa-slicer --align-xy 0,0 --center 100,100 --slice problem.stl --export-gcode --output out.gcode`
But it looks more or less like a hack than a cleaver solution.
There are also some experiments and comparisons of --center and --align-xy #7373
I use both cause:
 --align-xy doesn't respect X,Y coordinate to me... 
--center doesnot work to this problem when stands alone

el-bart, 2023-01-12:
yes, that one looks related, indeed.

thinking out loud here - it might be possible, that boundary checks ("is model outside of bed?") are done in a wrong place. i.e. instead of being done once, after all repositioning and coordinates transformation took place, it might be added around the particular parameter parsing. in that case checks would not always be executed and / or would return inconsistent results, depending on which options are passed. :thinking: 

since this is 100% reproducible, and exception is raised, `gdb`'s `catch throw` might come in handy here. :)

closed 2026-09-29 by prusaslicer-bot[bot]